hudbrog/gCodeViewer #32: Running converted PyCAM Gcode   on gCodeViewer

HardRainbow, 2015-03-14:
Dear Hudbrog :-)

I got some modified PyCAM Gcode i want to run with your gCodeViewer. Its converted to run on Marlin Firmware. Dont ask my why but i´d like to :-). Here a short example:

G90 (disable incremental moves)
G21 (metric)
F200.00000
G1 Z1.00000 F800
G14 P3 (wait for 3 seconds) F1000
G1 X39.850 Y23.488 F1000
G1 Z0.00000 F800
G1 X39.757 Y23.520 F150
G1 X39.675 Y23.549 F150
G1 X39.602 Y23.576 F150
G1 X39.538 Y23.599 F150
G1 X39.483 Y23.621 F150
G1 X39.436 Y23.642 F150
G1 X39.395 Y23.661 F150
G1 X39.362 Y23.679 F150
G1 X39.334 Y23.696 F150
G1 X39.294 Y23.731 F150
G1 X39.271 Y23.768 F150
G1 X39.259 Y23.810 F150
G1 X39.253 Y23.859 F150
G1 X39.251 Y23.901 F150

...

G1 X94.542 Y3.592 F150
G1 X94.569 Y3.593 F150
G1 X94.597 Y3.603 F150
G1 X94.649 Y3.634 F150
G1 X94.690 Y3.660 F150
G1 X94.738 Y3.690 F150
G1 X94.790 Y3.723 F150
G1 X94.846 Y3.759 F150
G1 X94.902 Y3.795 F150
G1 X94.959 Y3.832 F150
G1 X95.014 Y3.868 F150
G1 Z1.00000 F800
G14 P3 (wait for 3 seconds) F1000
G1 Z1.00000 F800
; FINISH THE CUT BY MOVING BIT TO SAFER POSITION F1000
G1 Z18.000000 ; MOVE BIT ABOVE PIECE F800
G1 X0.000000 Y0.000000 ; HOME X AND Y F150

What is missing for the interpreter of your genius algorithm to visualize my nasty GCode?
Just say "forget it" if its to difficult to explain ;-)

Greetings
Hardy

hudbrog, 2015-06-01:
Hi. 

Sorry, I know a lot of time has gone, but anyway. That viewer is really 3d-printer specific and because of that it expects gcode to have at least some extrusion (i.e. move along E/A/B/C axes). 

One way to trick it is to introduce M101 command when you're supposed to start milling and M103 when you stop (better if there are not Z moves in between).

closed 2015-06-01 by hudbrog

HardRainbow, 2015-06-02:
Thats awesome, thank you for that great hind :-) , i will alter my converter for supporting this "feature" :-)

Thanks alot!

HardRainbow, 2015-06-21:
Added the modification in github, also:

https://github.com/HardRainbow/CNC_gcode-to-reprap-converter