hudbrog/gCodeViewer #36: Wrong flow rate colors?

alexrj, 2015-05-31:
Hello @hudbrog!

I noticed that gCodeViewer calculates wrong flow rate values. Also, it displays more items in the legend than needed!  See the following small G-code file:

```
; generated by Slic3r 1.2.8-dev on 2015-05-31 at 21:42:31

M104 S200 ; set temperature
G28 ; home all axes
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G1 Z0.650 F7800.000 ; move to next layer (1)
G92 E0 ; reset extrusion distance
G1 X109.750 Y109.750 F7800.000 ; move to first perimeter point
G1 X90.250 Y109.750 E22.94318 F4800.000 ; perimeter ; dXY = 19.500mm ; dE = 0.36052mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45483mm^3/sec
G1 X90.250 Y90.250 E23.30371 ; perimeter ; dXY = 19.500mm ; dE = 0.36053mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45512mm^3/sec
G1 X109.750 Y90.250 E23.66423 ; perimeter ; dXY = 19.500mm ; dE = 0.36052mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45483mm^3/sec
G1 X109.750 Y109.675 E24.02336 ; perimeter ; dXY = 19.425mm ; dE = 0.35913mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45474mm^3/sec
G1 Z0.850 F7800.000 ; move to next layer (2)
G92 E0 ; reset extrusion distance
G1 X109.750 Y109.750 F7800.000 ; move to first perimeter point
G1 X90.250 Y109.750 E22.94318 F4800.000 ; perimeter ; dXY = 19.500mm ; dE = 0.36052mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45483mm^3/sec
G1 X90.250 Y90.250 E23.30371 ; perimeter ; dXY = 19.500mm ; dE = 0.36053mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45512mm^3/sec
G1 X109.750 Y90.250 E23.66423 ; perimeter ; dXY = 19.500mm ; dE = 0.36052mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45483mm^3/sec
G1 X109.750 Y109.675 E24.02336 ; perimeter ; dXY = 19.425mm ; dE = 0.35913mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45474mm^3/sec
```

These are just 2 layers with 5 extrusions each. All extrusions should be displayed with the same flow rate (10.45mm^3/sec).
However the first layer is rendered with many items in the legend (where do they come from?):

[image]

The second layer is rendered with a reasonable number of items in the legend (despite they could be grouped - see #28) but still the value seem to be off.

Thank you for your work :)

alexrj, 2015-05-31:
Oh, I forgot to say that the file was generated with `filament_diameter = 3` and that I set filament diameter to 3 manually in the *Printer Info* tab of gCodeViewer.

hudbrog, 2015-06-01:
Hi, Alessandro!

Actually, there were a couple of issues. I was not able to reproduce multiple colors issue you had (I do suspect that it comes from an older file you've loaded, and implemented a quick fix to make sure that color legend is updated properly). As for the speed calculation - you are absolutely right, it was not being calculated properly. I have fixed that as well. 
Your file does have an error though. For the first line of actual extrusion:
```
G92 E0 ; reset extrusion distance
G1 X109.750 Y109.750 F7800.000 ; move to first perimeter point
G1 X90.250 Y109.750 E22.94318 F4800.000 ; perimeter ; dXY = 19.500mm ; dE = 0.36052mm ; dE/XY = 0.01849mm/mm; volspeed = 10.45483mm^3/sec
```

You specify that dE is 0.36 but it's actually 22.94. Other calculations are correct.

Anyway, I've updated gcode.ws, so have a try and comment if there are still problems there. 
Thanks!

hudbrog, 2015-06-01:
btw, you can add a commet like:
; filament_diameter = 3.0
to the begining of the file so you won't need to enter it manually.