hudbrog/gCodeViewer #43: Spiral vase mode oddity - layer view only shows corners?

JBKingdon, 2016-04-05:
I sliced a 160x160x16mm shape in spiral vase mode for an extrusion test and wanted to check the extrusion rate in gCodeViewer. When I select the layer 2D view and scroll through the layers I only seem to get plots for the corners and nothing on the straight sections between. I suspect I'm missing something obvious, but can't think what!

zipped g-code file:

[big-rounded-testcube.gcode.zip](https://github.com/hudbrog/gCodeViewer/files/204404/big-rounded-testcube.gcode.zip)

Wholistic, 2017-05-26:
I had a similar glitch with a spiral vase mode file. Looks completely broken. 

[2016-05-26 - labyrinth.gcode.zip](https://github.com/hudbrog/gCodeViewer/files/1031448/2016-05-26.-.labyrinth.gcode.zip)

DrLex0, 2017-12-18:
gCodeViewer does not seem to be aware of vase mode, it treats every different Z coordinate in the file as a different layer, therefore you only see single points… Not sure how this can be solved, maybe allow a layer height override so it can group all print moves with a Z coordinate within e.g. 0.2mm, if that's your spiral vase Z resolution.