hudbrog/gCodeViewer #47: tiny gcode visualizer lib for tiny webserver

mkeyno, 2017-02-12:
sorry to open ticket for my question but actually I'm looking for light gcode visualizer or probably cad2gcode convertor in my webpage , but this webpage hosted on tiny ESP826612F module which is connected to Arduino GRBL gcode interpreter by serial line, so I should have great care about its weight , My plan is just connect to wifi module and open the browser and do designated CAM job.,As I'm not pro in client side script like JavaScript I was hopping could have good advise here, do you suggest to make some plugin from such library so I shouldn't have to download or call all library ?
thanks

SparkyCola, 2017-02-12:
You want to slice or generate CNC-paths with JS (which is the only well-supported client script)? I'm pretty shure there's no implementation of this out there since I can't think of a use-case where an alternative solution like a (native) program on pc/server couldn't be used, it would also take ages and a ton of script to do this (which would need to be hosted on another server which is not your device). You could run something on a real server (at least a Raspberry Pi) though, but there's still going to be quite some delay. There might also be some online services for cad2gcode available that have a JS-API.
About the viewer: I don't know about the capabilities of your Arduino, but running gCodeViewer on a server is really simple, just clone it, but if your performance is that low you might wanna remove the icons or replace them with vector items and compress the JS.

mkeyno, 2017-02-12:
thanks @SparkyCola  , my problem is most library which are visualizer gcode files or even parse vector image to gcode has lot of dependency files which is out of my module power so I'm looking for light lib or use some plugin so put most of this weight on browser side

SparkyCola, 2017-02-12:
Isn't gCodeViewer pretty light weight and only client-sided (besides transmission), i can only see JS, CSS and of course HTML and two images, compressing the code (I don't really know how to that, but stackoverflow or some tutorial might help there) is should reduce everything to a few KB that shouldn't kill your setup.

hudbrog, 2017-02-12:
Yeah, just as SparkyCola said. gCodeViewer has no server side whatsoever, all you need is just to serve files to the client's browser. Quickly looking over the code, it's quite easy to make it fit under 1M unpacked (minified though). esp8266 with something like https://github.com/me-no-dev/ESPAsyncWebServer and storing everything in flash fs should handle that with no problem at all.

SparkyCola, 2017-02-12:
Shouldn't there also be a way to remove not-required functions from angular js and other libs or get those from somewhere on the net or another "real" server? I mean would reduce everything to a few bits or at least KB.

mkeyno, 2017-02-12:
thanks @hudbrog  actually my webserver is the "ESPAsyncWebServer" , but the problem is number of file dependency is lot and make webserver  unstable and should consider, there  is no internet connection and module is  in access point mode, in this situation we gonna merge files in to the one, but still zipped file size is high and in many situation it is not fully loaded  in client browser so I was thinking maybe it is good idea to make plugin from library so put the weight on browser and only call class instance in the program, 
I might add I'm not good in JS and maybe not describe  problem well