hudbrog/gCodeViewer #48: Does not work for milling files.  Gives no error message.

jimls, 2017-03-28:
I now realize that this isn't what this program was made for but it isn't clearly noted it is for 3d printing.  gcode is used for lots of things.  Tried to use this on gcode for a milling project - was looking for something quick.  Progress indicators were two lines showing 100% each in green.  But I didn't get anything but a listing of the gcode.

SparkyCola, 2017-04-01:
Is your G-CODE noted like `G 1` with the space between command and number, I've seen this when I used Inkscape to make G-CODE for my 3d-printer which I used as plotter.

svenstucki, 2017-06-29:
I wanted to visualize some gcode for a (3D printer based) laser engraver and have the same problem. Gcode uploads and can be viewed in the text pane, but nothing else.

In case anyone wants to play around a bit, here's some example gcode: https://pastebin.com/4EiDJiw8

This might also be related to #40, my example contains some G2/G3 codes. They were created by Inkscape with the ShinyLaser extension. The gcode generated by this used codes like G00, G01 and G02. I've tried replacing them (with G0, G1, G2) but it doesn't help.