hudbrog/gCodeViewer #49: Bug in renderer3d then also breaks 2D view

Clotoo, 2017-05-24:
**Description :**
When a GCODE contains an actual ZERO coordinate (as X, Y, or Z), then the corresponding point in the 3D visualizer doesn't appear where it should be. The bug doesn't appear at first in the 2D view, but after opening 3D view it appears, and when going back to 2D view it appears there too.

**Steps to reproduce :**
- Load this offending gcode : https://www.dropbox.com/s/y4bsxuew852eer9/batman_spinner.gcode?dl=0
- Try 2D view, works fine
- Move to 3D view, notice issue at the bottom (the bottom point of the triangle should be at X=0 but it's shifted)
- Move back to 2D view and refresh canvas by moving around, the point is now shifted there too.

**Cause :**
After thorough investigation, it appears the offending point (for the first layer) corresponds to line 513 in the GCODE
`G1 X-0.000 Y-17.597 E3.84787`
It is not a parsing problem though, the parser correctly reads it as `{ x=0  y=-17.597  z=NaN, ... }`
The problem is in the 3D model builder when iterating through the `cmds` (renderer3d.js line 49)
```
            if(!cmds[j].x)cmds[j].x=prevX;
            if(!cmds[j].y)cmds[j].y=prevY;
            if(!cmds[j].z)cmds[j].z=prevZ;
```
Number zero is interpreted as False.

**Fix :**
I am not familiar enough with your code to be sure, but I think replacing the `!cmds[j].x` with a `isNaN(cmds[j].x)` instead should do the job. `isNaN` will return true in case of `NaN` or `undefined`.

hudbrog, 2017-05-24:
That is THE BEST bug report I've ever seen. So good that I even applied the fix =) 
Works now =) Thanks a lot!

closed 2017-05-24 by hudbrog

fmargueirat, 2017-05-24:
I must agree. As a developer, I wish all the bug reports I get were these
detailed. Adding the potential solution is a great plus.