hudbrog/gCodeViewer #50: Allow loading of .gx files

btzy, 2017-06-30:
*.gx files are the output of FlashPrint, the FlashForge slicer.  Internally, .gx files are gcode files with a small header at the top of the file.  They load correctly into the gCode Viewer already, but since it refuses .gx files, I need to rename them to .g files before gCode Viewer would accept them.

Maybe, instead of refusing to load the file, a warning could be displayed, with the option to attempt loading anyway.

Here's a sample .gx file produced by FlashPrint:
[clip.zip](https://github.com/hudbrog/gCodeViewer/files/1115363/clip.zip)

PeterWBoyer, 2017-07-10:
I agree with btzy.  I too use FlashPrint which produces all my gcode files as ".gx" files.  The gCodeViewer ought to automatically recognize the .gx files and open to be viewed.

ChristopherHoffman, 2017-09-30:
Support for gx has been added! Give it a try ^_^

btzy, 2017-10-01:
Thank you! It works well :)

closed 2017-10-01 by btzy