hudbrog/gCodeViewer #53: gCodeViewer "skips" a layer

b-morgan, 2017-08-11:
The attached file generated with Cura Lulzbot Edition (21.0.4) and Cura Lulzbot Edition Beta (2.6.21) for a TAZ 6 printer skips layer 7 when displayed with the gCodeViewer. It happens in multiple browsers (Firefox, Edge, IE, Chrome). Turning off the "Sort layers by Z" option will sometimes eliminate the issue. Note that layer 8 is actually the gcode of layer 7 so the following layer numbers are off by one.

[Filament_clip_3mm_and_175.zip](https://github.com/hudbrog/gCodeViewer/files/1217873/Filament_clip_3mm_and_175.zip)

The client computer is Windows 10.

tsh-xx, 2017-10-08:
https://github.com/Ultimaker/Cura/issues/2565 has another testcase. The workaround above works. 

I observe OctoPrint has the same behaviour.

b-morgan, 2017-10-08:
OctoPrint is using the same code as the gCodeViewer (without access to "Sort layers by Z" which is always on).