hudbrog/gCodeViewer #56: Request: Total time spent to get to layer

Jikiya, 2017-10-03:
In the Layer view it would be nice to have a running total of time spent to get to said layer.  I personally like watching bridges being made, and always check to see what layer they'll be on.  Would be nice if I didn't need to calculate the time myself.

Thanks.

yoyo42, 2019-05-11:
This would also be useful when I know there's something like a filament change coming up and need to know how long I have got before needing to come back to the printer.

It might also be nice if the previous calculated/analysed times were kept track of and either fed back to the authors to help improve the algorithm, or even just tracked to help see how accurate the estimates are likely to be.

bechet-john, 2020-09-08:
Is there a new version that includes accumulative time when reaching certain levels
Many thanks

John
Spain