hudbrog/gCodeViewer #69: Gcode files with no extrusion in them will not render

zsoltmar, 2018-05-03:
Love this viewer! I'm trying to use it for a laser engraving software where there are only G0 and G1 commands, no E for extrusion for obvious reasons. Is there a setting I'm missing that needs to be turned on in this scenario?
To reproduce, here's an example Gcode: 
```G0 X135.000004 Y105 F5400
G1 X134.921151 Y103.746667 F300
G1 X134.685836 Y102.5131 F300
G1 X134.297769 Y101.318752 F300
G1 X133.763071 Y100.182461 F300
G1 X133.090173 Y99.122144 F300
G1 X132.289689 Y98.154525 F300
G1 X131.374243 Y97.294864 F300
G1 X130.35827 Y96.556717 F300
G0 X115.851093 Y105.000125 F5400
G1 X125.001469 Y114.150502 F300
G1 X134.151846 Y105.000125 F300
G1 X125.001469 Y95.849748 F300
G1 X115.851093 Y105.000125 F300```

Thank you

zsoltmar, 2018-05-03:
As a workaround, I add an E value to each G1 movements increasing every step. Once the code loaded into the viewer I then run a function that parses all the lines and removes the E values from the gcode. Not super efficient but at least it works until a proper fix is set in place. Cheers!

Wingman4l7, 2020-07-03:
Yeah, ran into this issue pretty quickly -- makes the viewer unfortunately pretty useless for gcode-based pen plotters. 😕