hudbrog/gCodeViewer #86: Feature request: when the sliders are at a certain line of a certain layer, allow to identify the location in the gcode file.

zapta, 2020-08-08:
The use case is like that:

1. User wants to understand a defect in the printing.
2. User open the gcode file in the viewer and move the layer and line sliders to the point of defect.
3. User wants to find the corresponding location in the gcode to understand what causes the problem.

For example, when switching to the gcode view, highlight the current line, not just the code bloc with current layer.

planetar, 2022-07-04:
I came to this place to write a Feature Request just like this. To have a pointer in the GCODE to the position corresponding to the current nozzle position would be very helpful to any manual editing of GCDE, like:
- insert M600 at specific positions for filament change/multicolor 
- identify where the contour ends and the filling starts so I can create some 2D-artwork with the printer
- learn more about GCODE

However, the repo looks rather dormant...

hudbrog, 2022-07-04:
It already works like that. Move slider to select the layer, move horizontal slider to only show the lines to to the moment you need, switch to gcode tab - beginning of that layer up to the part selected by horizontal slider is chosen, so the first unselected line is what you are looking for.

planetar, 2022-07-04:
Wow! an instant reply. Thanks a lot, I see it now