hudbrog/gCodeViewer #88: Question: Is possible to run gCodeViewver locally without docker?

lucky62, 2021-01-07:
Is possible to run gCodeViewer locally without docker?
If yes, then how?

hudbrog, 2021-01-07:
Yes, but you should not do it )
Basically, you will need to run your browser with a command line key that would disable local file access restrictions: `--allow-file-access-from-files` is case of chrome. It's unsafe and is not recommended.

closed 2021-01-07 by hudbrog

lucky62, 2021-01-07:
When gCodeViewer is running then it is acting as a server listening on port 8080.
I would like to run it by the same way but I don't want to install docker...

Findus23, 2021-01-07:
Of course, you could also run a local webserver (e.g. something like `python3 -m http.server 1234`) and then access the site in your webbrowser.

lucky62, 2021-01-07:
> Of course, you could also run a local webserver (e.g. something like `python3 -m http.server 1234`) and then access the site in your webbrowser.

Many thanks. Now I understand that this is a pretty web page which can be served by any regular web server.